YUKAI/konashi-android-sdk #3: konashi（無印）に対応させる

e10dokup, 2015-04-03:
* 取り敢えずUUIDを2クラスに分離させる？
* findしたkonashiのネームで取得するクラス変えたらいんじゃね感
* konashi（初代）のBASE_UUIDください

e10dokup, 2015-04-04:
どうやらkonashi1だけBluetooth SIG標準らしいのでBASE_UUIDは ”00000000-0000-1000-8000-00805f9b34fb”らしい

e10dokup, 2015-04-04:
よくよく考えたらHashMap使えばちょっといいかんじに実装できそう

izumin5210, 2015-04-05:
定数にHashMapはどうなんだろう…
Enumつかうのが無難な気が

izumin5210, 2015-04-05:
たとえば`KonashiUUID`を`enum`にしておいて，下記のようなUtilityメソッドを定義するとか．

```java
KonashiUtils.getUUID2(KonashiUUID.BATTERY_SERVICE)
// => 229B180F-03FB-40DA-98A7-B0DEF65C2D4B

KonashiUtils.getUUID1(KonashiUUID.BATTERY_SERVICE)
// => ????????-03FB-40DA-98A7-B0DEF65C2D4B
```

e10dokup, 2015-04-06:
@izumin5210 なるほど強い

izumin5210, 2015-06-11:
- [x] initialize
- [x] find
- [x] disconnect
- [x] isConnected
- [x] peripheralName
- [x] pinMode
- [x] pinPullup
- [x] digitalRead
- [x] digitalWrite
- [x] analogReadRequest
- [x] analogRead
- [x] pwmMode
- [x] pwmPeriod
- [x] pwmDuty
- [x] uartMode
- [x] uartBaudrate
- [x] uartWrite
- [ ] i2cMode
- [ ] i2cStartCondition
- [ ] i2cStopCondition
- [ ] i2cWrite
- [ ] i2cReadRequest
- [ ] i2cRead
- [ ] reset
- [ ] batteryLevelReadRequest
- [ ] batteryLevelRead
- [ ] signalStrengthReadRequest
- [ ] singalStrengthRead

izumin5210, 2015-08-09:
一旦close（また気が向いたらやる）

closed 2015-08-09 by izumin5210