YUKAI/konashi-android-sdk #4: Konashi v1/v2を（無意識的に）切り替え可能にする

izumin5210, 2015-07-02:
#3 関連

## WHY
開発者（SDK利用者）にKonashiのバージョンを意識させるのはエレガントではない．
どうにかして自動識別できたほうが当然ながら使い勝手は向上する．

## HOW

### konashiの接続フロー

> `deviceName.startWith(KONASHI_DEVICE_NAME) == true`（KonashiBaseManageのconstants）ならばリストに表示
>
>  * KonashiBaseManager.java#L350付近，`#onLeScan()`
>  * `KONASHI_DEVICE_NAME`はv1が`konashi`，v2は`konashi2`

この`device.getName()`を利用すればバージョン識別が可能っぽい？

> `KonashiBaseManager#connect()`でdeviceを`mConnectedDevice`とかに保存して
> `mConnectedDevice.getName()`からv1/v2を返すメソッドをKonashiBMに実装したら

> ```java
> enum KonashiVersion {
>     V1, V2;
> }
> ```
> みたいなのを定義しておけば拡張性も高い

sagiii, 2015-07-29:
ありがとうございます。

ios-sdkでは、software_revision_stringでバージョンの切り替えを行っています。（基本的にはkonashiのファームウェアバージョンに対応しており、konashi2は2.x.xとすることとしています）

izumin5210, 2015-07-30:
https://github.com/YUKAI/konashi-ios-sdk/blob/275f3804be6b96ac4c56859f48b320f01fb9e946/Konashi/Konashi/Impl/KNSPeripheralBaseImpl.m#L20

sagiii, 2015-07-30:
あるいは、
https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicViewer.aspx?u=org.bluetooth.characteristic.software_revision_string.xml
が本家になります。

izumin5210, 2015-08-09:
一旦close（また気が向いたらやる）

closed 2015-08-09 by izumin5210