YUKAI/konashi-android-sdk #8: pwmPeriod, pwmDuty の間に Thread.sleep を入れないと安定しない

kiryuxxu, 2015-07-29:
```java
mKonashiManager.pwmPeriod(mPinNumber, period);
mKonashiManager.pwmDuty(mPinNumber, duty);
```

上記のように `pwmPeriod` と `pmwDuty` を連続して実行する場合、`Thread.sleep` を間に入れないと動作が安定しません。

sagiii, 2015-07-29:
@kiryuxxu と話して、sdkにsleepを（100は長い気もしますが）入れるでもいいかなと思っています。（ios-sdkもi2c周りですが１箇所入ってます）

izumin5210, 2015-07-29:
Androidアプリケーションにおいて，UI Threadをブロックするのはバッドプラクティスになります（実際，ある程度以上ブロックされたらExceptionが投げられる）．
イベントが発行されるスレッドを選択できない以上はドキュメント上で注意喚起するのがいいと思われますが，如何でしょうか．

izumin5210, 2015-07-29:
SDKがユーザに無断で`Thread.sleep()`するより，`KonashiObserver#onUpdatePwmPeriod()`のようなコールバックや`KonashiObserver#onUpdate(KonashiEvent)`，`KonashiEvent.UPDATE_PWM_PERIOD`といったイベントを用意してあげてあとはユーザに任せるという解決策もアリだと思います

izumin5210, 2015-08-09:
#42 にてPWM Duty/Period変更のコールバックを追加したので，本Issueはcloseします．

closed 2015-08-09 by izumin5210