YUKAI/konashi-android-sdk #76: pwmをLEDモードで有効化した際にonUpdatePwmModeが動かない

e10dokup, 2015-08-19:
https://github.com/YUKAI/konashi-android-sdk/blob/309cbade2fccae8fdca040084ec1c8450c0cbae0/konashi-android-sdk/src/main/java/com/uxxu/konashi/lib/KonashiManager.java#L366-L389

## what

PWM_ENABLE_LED_MODEをmodeに与えて``` pinMode ```を実行するとkonashiに``` pwmPeriod,  pwmLedDrive, addWriteMessage ```で3つのメッセージが連続して送信され，結果的に後ろに続く``` addWriteMessage ```が無視されるような挙動をしたり，安定しなくなる．その結果pwmの有効化がなされず，コールバック（``` onUpdatePwmMode ```）も受け取れなくなっている．

なお無効化する場合は``` addWriteMessage ```のみを実行しないのでちゃんと無効化される

## why

https://github.com/YUKAI/konashi-android-sdk/pull/39

この時に取り去ったFiFoTimerが勝手に遅延をかましていたので連続的に実行しても気づかなかっただけらしい．実質的にはいつぞやの「pwmDutyとpwmPeriodの間にsleep入れないと安定しない」あれと原因は一緒っぽい．つまりkonashiのファームウェア由来かと

kiryuxxu, 2015-08-20:
Promise や Job Queue のようなものを使って、前の処理が確実に終わってから次の処理を実行できるインタフェースが欲しいですね。以下、こんな感じに書けたら綺麗かもというアイデアです。

```java
final int targetPin = 0;
final int value = 123;
konashi.createTaskWithPin(targetPin)
    .pwmPeriod(Konashi.PWM_LED_PERIO)
    .pwmLedDrive(0.0F)
    .execute((Task task) -> {
        if (task.isFailed()) {
            Log.d(TAG, "Error", task.getError());
            return;
        }
        konashi.addWriteMessage(KonashiUUID.PWM_CONFIG_UUID, val);
    });
```

izumin5210, 2015-08-20:
いまどきコールバック前提のAPIはつらさしかないですもんね…
検討の価値ありそうです:+1:

izumin5210, 2015-08-22:
Promisifyするのはｺｽﾄ高そう&&外部依存が発生しそうなので，別Issueに

izumin5210, 2015-09-30:
#82 APIをPromise化したので多分大丈夫

closed 2015-09-30 by izumin5210