YUKAI/konashi-android-sdk #80: Konashi1.0と2.0でPWMの対応ピンが違う

e10dokup, 2015-08-22:
多分他にもこの手の何かがあるはず

* 1.0：0~7
* 2.0：0~2

sagiii, 2015-08-23:
http://konashi.ux-xu.com/documents/
の黄色い箇所がkonashi1.0とkonashi2.0の差分になります。
ご確認ください。

izumin5210, 2015-09-30:
#82 にて，PIO0〜PIO2以外を受け付けないようにする予定（暫定対応）

kiryuxxu, 2015-09-30:
入力チェックの補助として `@IntDef` や `@ IntRange` を使うというのはどうでしょうか?

sagiii, 2015-09-30:
konashi1とkonashi2の差分に関しまして、

ダイジェストといたしましては、
https://www.value-press.com/pressrelease/134384
こちらの表１が差分となります。

詳細といたしましては、
http://konashi.ux-xu.com/documents/
こちらのページの、"konashi 2"で検索してハイライトされる箇所になります。

鷺坂

-- 
鷺坂 隆志
SAGISAKA Takashi
sagisaka@ux-xu.com
090-5769-0963

ユカイ工学株式会社
http://www.ux-xu.com/

東京都新宿区富久町16-11 武蔵屋スカイビル101号
http://goo.gl/maps/tuv8z

TEL:03-6380-4710

【新製品】BOCCO 〜 家族をつなぐコミュニケーションロボット 〜
http://youtu.be/qiWEhdBeQmA

izumin5210, 2015-09-30:
@kiryuxxu それも考えたのですが，SDK利用者がsupport.annotationsを利用していた場合にexcludeする必要が出てくるためちょっと厄介かな，ということで悩んでました

もしくはKonashiクラス内の定数をEnumにしてしまうのもありかなと思ってます
すべてEnumにしてしまえば，konashi1/konashi2対応のときも利用するEnumクラスを差し替えればいいだけになりますので…
こちらのデメリットとしては，影響範囲が大きいため実装に時間を取られてしまう可能性がある点です

kiryuxxu, 2015-09-30:
@izumin5210 確かにそれは悩ましいですね。
クリティカルな部分ではないので将来的に Enum 化するとかで良いと思います。

izumin5210, 2015-09-30:
ではこのIssueは一旦閉じて，別IssueとしてEnum化をあげておきます

closed 2015-09-30 by izumin5210