YUKAI/konashi-android-sdk #94: test_all_functionsのUART sendを連打するとクラッシュする

e10dokup, 2015-09-28:
#82 

UARTでSENDボタンを連打していると
https://github.com/YUKAI/konashi-android-sdk/blob/feature/bletia/konashi-android-sdk/src/main/java/com/uxxu/konashi/lib/dispatcher/CharacteristicDispatcher.java#L31-L36
にてクラッシュするっぽい

``` 
09-29 01:16:45.727  25939-25939/com.uxxu.konashi.test_all_functions E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: com.uxxu.konashi.test_all_functions, PID: 25939
    java.lang.IllegalArgumentException
            at com.uxxu.konashi.lib.dispatcher.CharacteristicDispatcher.valueOf(CharacteristicDispatcher.java:35)
            at com.uxxu.konashi.lib.dispatcher.CharacteristicDispatcher.onDone(CharacteristicDispatcher.java:40)
            at com.uxxu.konashi.lib.dispatcher.CharacteristicDispatcher.onDone(CharacteristicDispatcher.java:14)
            at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:107)
            at org.jdeferred.impl.AbstractPromise.done(AbstractPromise.java:63)
            at org.jdeferred.impl.AbstractPromise.then(AbstractPromise.java:166)
            at com.uxxu.konashi.lib.KonashiManager.execute(KonashiManager.java:531)
            at com.uxxu.konashi.lib.KonashiManager.uartWrite(KonashiManager.java:395)
            at com.uxxu.konashi.test_all_functions.CommunicationFragment$3.onClick(CommunicationFragment.java:109)
            at android.view.View.performClick(View.java:4803)
            at android.view.View$PerformClick.run(View.java:19981)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5430)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:913)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:706)
```

izumin5210, 2015-09-29:
`UartDispatcher`さんがUART_WRITEのUUIDを知らないからうまくハンドリングできなくなって落ちてる．
`uartWrite()`のあとに`mUartDispatcher`を挟まなければ大丈夫なはず:bulb:
https://github.com/YUKAI/konashi-android-sdk/blob/0accf61239e580f7df1ed7e4602808b3c3e7d4fc/konashi-android-sdk/src/main/java/com/uxxu/konashi/lib/KonashiManager.java#L403

e10dokup, 2015-09-29:
http://lgtm.in/p/SNHjiz8mq

closed 2015-09-30 by izumin5210