YUKAI/konashi-android-sdk #108: i2cWrite で書き込んだ最後の値が壊れる

kiryuxxu, 2015-10-01:
https://github.com/YUKAI/konashi-android-sdk/tree/feature/remove-debug-code-from-test-all-functions
の test_all_functions の I2C で `ABCDEFGHIJKLMNOP` を書き込むと最後の `P` が壊れているように見える。

## テスト内容

Konashi の I2C を Arduino で読むテストコードを作り、同時にロジアナでも値を確認しながら test_all_functions から `ABCDEFGHIJKLMNOP` を書き込む。

Arduino 側のコード
```
#include <Wire.h>

void setup() {
  Wire.begin(0x01f);
   Wire.onReceive(receiveEvent); // register event
  Serial.begin(9600); 
}

void loop() {
  delay(100);
}

void receiveEvent(int howMany) {
  while (1 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  int x = Wire.read();    // receive byte as an integer
  Serial.println(x);         // print the integer
}
```

Arduino の Serial output
```
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
ABCDEFGHIJKLMN79
```

ロジアナで確認したところ、`P` のデータが送出されてないようでした。

izumin5210, 2015-10-02:
#109 にて対応しました
確認よろしくお願いいたします

kiryuxxu, 2015-10-02:
Arduino 側のコードを少し変更して試したところ問題無さそうでした。

Arduino
```
#include <Wire.h>

void setup() {
  Wire.begin(0x01f);
   Wire.onReceive(receiveEvent); // register event
  Serial.begin(9600); 
}

void loop() {
  delay(100);
}

void receiveEvent(int howMany) {
  while (1 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  char x = Wire.read();    // receive byte as an integer
  Serial.println(x);         // print the integer
}
```

Results
```
ABCDEFGHIJKLMNO  // Konashi sent "ABCDEFGHIJKLMNO"
ABCDEFGHIJKLMNOP  // Konashi sent "ABCDEFGHIJKLMNOP"
ABCDEFGHIJKLMNOP  // Konashi sent "ABCDEFGHIJKLMNOPQ"
```

closed 2015-10-02 by kiryuxxu