YUKAI/konashi-android-sdk #142: release/1.0.1 の test_all_functions で UART を送信すると I2C の received data に謎の空白が混じる

kiryuxxu, 2015-10-13:
I2C と UART を ON にした状態で、UART から `ABCDEFGHIJKLMNOP` を送信すると 12 文字ぐらいの感覚で空白が混ざる。

[image]

e10dokup, 2015-10-13:
@kiryuxxu 手元のkonashiでTX-RX間を接続して同様のことをしましたが，再現しません．どのような環境上で発生したのか教えていただいてもよろしいでしょうか？

kiryuxxu, 2015-10-13:
[image]

こちらのテスト用 jig を使ってもらったうえで以下の操作を行ってみてください。

1. Konashi に接続
2. Communication を開く
3. UART を ON にする
4. I2C を ON にする
5. I2C の SEND ボタンを押す
6. UART に値が表示されたのを確認したら CLEAR ボタンを押して 5 を再度繰り返す

e10dokup, 2015-10-13:
@kiryuxxu ありがとうございます．手元にTAF用ボードが無いので同一環境上ではありませんが．現象が再現しました．

e10dokup, 2015-10-13:
@kiryuxxu 確認したところ，本件については

* そもそもUART_RX_NOTIFICATION_UUIDから端末に送られるデータが途中で分断される（おそらく送信側がchar 1文字づつ送っていることに起因）
* UARTのReadの受信フォーマットが byte列length，本来のbyte列で，表示ではlengthの値も表示しようとしている（制御文字なので空白になる）

が原因だと思います．対策として，Readの受信の値からlengthを取り除くようにし，I2CのReadと同じような形にします．

sagiii, 2015-10-13:
対策のご提案をいただき、ありがとうございます！

UART_RX_NOTIFICATION_UUIDからは、マイコン内でのバッファを最小化するために長くない間隔でバッファ内のデータをnotifyしておりますので、前者はkoshianの仕様とお考えいただければ幸いです。お手数をおかけしてすみません。。
（正確には、18バイト以下のすでに到達しているデータが随時送信されてきます）

-- 
鷺坂 隆志
SAGISAKA Takashi
sagisaka@ux-xu.com
090-5769-0963

ユカイ工学株式会社
http://www.ux-xu.com/

東京都新宿区富久町16-11 武蔵屋スカイビル101号
http://goo.gl/maps/tuv8z

TEL:03-6380-4710

【新製品】BOCCO 〜 家族をつなぐコミュニケーションロボット 〜
http://youtu.be/qiWEhdBeQmA

e10dokup, 2015-10-13:
@sagiii ありがとうございます．この対策で進めていこうと思います

closed 2015-10-16 by e10dokup