YUKAI/konashi-android-sdk #153: SPI support

izumin5210, 2015-11-14:
* [ ] enable/disable
* [ ] speed
* chip select -> digital write
* [ ] spi write
* [ ] spi read (read request?)
  - notification

ZeroCrossroad, 2015-11-14:
SPI speed
 KonashiSPISpeed200K = 20,
 KonashiSPISpeed500K = 50,
 KonashiSPISpeed1M = 100,
 KonashiSPISpeed2M = 200,
 KonashiSPISpeed3M = 300,
 KonashiSPISpeed6M = 600, 

SPI mode
 KonashiSPIModeEnableCPOL0CPHA0 = 0,
 KonashiSPIModeEnableCPOL0CPHA1 = 1,
 KonashiSPIModeEnableCPOL1CPHA0 = 2,
 KonashiSPIModeEnableCPOL1CPHA1 = 3,
 KonashiSPIModeDisable = -1, 

SPI Endianness
 KonashiSPIBitOrderLSBFirst = 0,
 KonashiSPIBitOrderMSBFirst = 1,

ZeroCrossroad, 2015-11-14:
UUIDs

KonashiSPIData (UUID = 229B3017-03FB-40DA-98A7-B0DEF65C2D4B) 
length: variable length (Max length 64byte)

KonashiSPINotification (UUID = 229B3018-03FB-40DA-98A7-B0DEF65C2D4B) 

KonashiSPIConfig (UUID = 229B3016-03FB-40DA-98A7-B0DEF65C2D4B)
length: 4byte
byte[0]  = SPI_MODE
byte[1] = BigEndian,LittleEndian
byte[2] = SPISpeed >> 8;
byte[3] = SPISpeed & 0xff;

closed 2015-12-24 by e10dokup