YUKAI/konashi-android-sdk #154: uartWrite(String) で  java.lang.IllegalArgumentException が発生する

kiryuxxu, 2015-11-20:
`uartWrite(byte[])` だと発生していないので実装漏れ?

```
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime: java.lang.IllegalArgumentException
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at com.uxxu.konashi.lib.dispatcher.CharacteristicDispatcher.valueOf(CharacteristicDispatcher.java:35)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at com.uxxu.konashi.lib.dispatcher.CharacteristicDispatcher.onDone(CharacteristicDispatcher.java:40)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at com.uxxu.konashi.lib.dispatcher.CharacteristicDispatcher.onDone(CharacteristicDispatcher.java:14)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at org.jdeferred.android.AndroidDeferredObject$InternalHandler.handleMessage(AndroidDeferredObject.java:82)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at android.os.Handler.dispatchMessage(Handler.java:102)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at android.os.Looper.loop(Looper.java:135)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at android.app.ActivityThread.main(ActivityThread.java:5254)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at java.lang.reflect.Method.invoke(Native Method)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at java.lang.reflect.Method.invoke(Method.java:372)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
11-20 17:06:51.020 22165-22165/jp.co.ntt.betobetosan.android E/AndroidRuntime:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
```

kiryuxxu, 2015-11-20:
内部の実装が違う様です。

https://github.com/YUKAI/konashi-android-sdk/blob/master/konashi-android-sdk/src/main/java/com/uxxu/konashi/lib/KonashiManager.java#L433-L443
```
    public Promise<BluetoothGattCharacteristic, BletiaException, Void> uartWrite(byte[] bytes) {
        return execute(new UartWriteAction(getKonashiService(), bytes, mUartStore));
    }

    /**
     * UART でデータを送信する
     * @param string 送信するデータ(string)
     */
    public Promise<BluetoothGattCharacteristic, BletiaException, Void> uartWrite(String string) {
        return execute(new UartWriteAction(getKonashiService(), string, mUartStore), mUartDispatcher);
    }
```

izumin5210, 2015-11-30:
#148 にて対応しました

closed 2015-11-30 by izumin5210