YUKAI/konashi-android-sdk #161: FWバージョン2.1.3以下へのKonashiへの対応

e10dokup, 2015-12-13:
# What
FW ver.2.1.3でSPIが追加されたことにより，Characteristicが増加した．それによってFW ver.2.0.0のままのKonashiに接続しようとするとCallbackHandlerでNPEを出し，接続が取れない

https://github.com/YUKAI/konashi-android-sdk/blob/develop/konashi-android-sdk/src/main/java/com/uxxu/konashi/lib/CallbackHandler.java#L48-L58

kiryuxxu, 2015-12-15:
こちら FW 2.0.0, 2.1.3 共に再現しました。

```
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt: java.lang.NullPointerException
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.action.CharacteristicAction.<init>(CharacteristicAction.java:16)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.action.EnableNotificationAction.<init>(EnableNotificationAction.java:19)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.Bletia.enableNotification(Bletia.java:109)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at com.uxxu.konashi.lib.CallbackHandler.onServicesDiscovered(CallbackHandler.java:55)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.EventEmitter.emitServiceDiscovered(EventEmitter.java:43)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.Bletia.onServiceDiscovered(Bletia.java:132)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.BluetoothGattCallbackHandler.onServicesDiscovered(BluetoothGattCallbackHandler.java:40)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at info.izumin.android.bletia.wrapper.BluetoothGattCallbackWrapper.onServicesDiscovered(BluetoothGattCallbackWrapper.java:20)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at android.bluetooth.BluetoothGatt$1.onSearchComplete(BluetoothGatt.java:299)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:215)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at android.os.Binder.execTransact(Binder.java:404)
12-15 16:15:05.596 23815-23827/com.uxxu.konashi.test_all_functions W/BluetoothGatt:     at dalvik.system.NativeStart.run(Native Method)
```

`bletia.enableNotification(service.getCharacteristic(KonashiUUID.SPI_NOTIFICATION_UUID), true);` をコメントアウトすると両バージョンで onConnect() が正常動作する事を確認しています。

sagiii, 2015-12-17:
fw 2.1.3および2.0.0のkonashiに接続でき、PIO, PWM, Analog, Konashi Infoが触れることを確認しました。

closed 2015-12-24 by e10dokup