YUKAI/konashi-android-sdk #176: Android 6 で正常に動作しない

kiryuxxu, 2016-01-18:
Android 6 からセキュリティモデルが変わり、BLE を利用するために `ACCESS_COARSE_LOCATION` もしくは `ACCESS_FINE_LOCATION` を明示的に追加する事が必要になりました。

参考
* http://developer.android.com/intl/ja/training/permissions/requesting.html
* http://developer.radiusnetworks.com/2015/09/29/is-your-beacon-app-ready-for-android-6.html

Android 6 で `startLeScan()` を行った場合のエラーメッセージは以下のとおりです。
```
01-18 14:13:09.134 19659-19659/com.uxxu.konashi.inspector.android D/BluetoothAdapter: stopLeScan()
01-18 14:13:09.136 19659-19659/com.uxxu.konashi.inspector.android D/BluetoothAdapter: scan not started yet
01-18 14:13:09.137 19659-19659/com.uxxu.konashi.inspector.android D/BluetoothAdapter: startLeScan(): null
01-18 14:13:09.140 19659-19659/com.uxxu.konashi.inspector.android D/BluetoothAdapter: STATE_ON
01-18 14:13:09.147 19659-19672/com.uxxu.konashi.inspector.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder: Caught a RuntimeException from the binder stub implementation.
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder: java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder:     at android.os.Parcel.readException(Parcel.java:1599)
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder:     at android.os.Parcel.readException(Parcel.java:1552)
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder:     at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:772)
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder:     at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:324)
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder:     at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
01-18 14:13:09.159 19659-19672/com.uxxu.konashi.inspector.android W/Binder:     at android.os.Binder.execTransact(Binder.java:453)
```

上記以外の部分でも正常に動作しない箇所があるため Android 6 へ対応するブランチを切った方が良いかもしれません。

sagiii, 2016-03-22:
Android 6 で notification を有効にし終えた通知が来ない、という問題があるようです。

正常な場合：
```
setCharacteristicNotification() - uuid: 229b3003-03fb-40da-98a7-b0def65c2d4b enable: true
...
onDescriptorWrite() - Device=08:00:42:F0:1F:7A UUID=229b3003-03fb-40da-98a7-b0def65c2d4b
```
というようにdescriptorへのwriteが完了したeventが発火するのですが、Android 6ではこのイベントが起きないためブロックしてしまうようです。

konashi側のdescriptorの実装にも問題がある可能性もあります。

e10dokup, 2016-08-05:
v2.0.0にて修正完了。closeします

closed 2017-06-06 by kiryuxxu